IntelLabs/kAFL #236: TNT decoder returned: decoder_error

M3noetius, 2023-09-14:
The fuzzer is working and reports coverage and finds the bug from the windows kernel example. 

When I perform `fuzz cov` I receive the following error coming from [[here]](https://github.com/nyx-fuzz/libxdc/blob/master/src/decoder.c#L930)  

My system has a i7-1360P and I do not know if intel has done some updates on the instructions or in general they have start using know an unknown or somewhat not used "pt-instruction" more? 

Is it possible to give me a hint on what might be the fault, so I can look for it ? The error is 90% coming from libxdc ? Should I check how Intel-PT works ?

Thank you ! 

Also check this commits from libipt [[here](https://github.com/intel/libipt/commit/c91ee4faeec3fde5440fcfca8b15bb418142a065)][[here](https://github.com/intel/libipt/commit/64780d0a6d2e1e303785663eae599857b3ae2587)]
  

```
0xfffff803444d0000	0xfffff803444d7000	\??\C:\Users\vagrant\Desktop\src\kafl_vulnerable_driver\x64\Debug\kAFL vulnerable driver\kAFLvulnerabledriver.sys
                                                                                                                                   herecess-1:   0%|                                                                                              | 0/6 [00:00<?, ?it/s]

Processing payload_00001..
/home/user/kAFL/kafl/qemu/nyx/auxiliary_buffer.c:175:9: runtime error: load of misaligned address 0x7fb488601182 for type 'volatile uint32_t', which requires 4 byte alignment
0x7fb488601182: note: pointer points here
 00 00  01 04 00 00 00 00 00 01  01 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  00 00 00 00 00 00
              ^ 
ptdump of payload_00001 failed with return code 1. Output:

ERR: 	TNT 11 at position <0xfffff803444d1574,0xfffff8033a02a6b5>
[*] decoder returned: decoder_error
[*] page fault addr:   	0x0

Processing payload_00002..
ptdump of payload_00002 failed with return code 1. Output:

ERR: 	TNT 12 at position <0xfffff803444d1574,0xfffff8033a02a6b5>
[*] decoder returned: decoder_error
[*] page fault addr:   	0x0

Processing payload_00003..
ptdump of payload_00003 failed with return code 1. Output:

ERR: 	TNT 12 at position <0xfffff803444d1574,0xfffff8033a02a6b5>
[*] decoder returned: decoder_error
[*] page fault addr:   	0x0

Processing payload_00004..
ptdump of payload_00004 failed with return code 1. Output:

ERR: 	TNT 9 at position <0xfffff803444d1574,0xfffff8033a02a6b5>
[*] decoder returned: decoder_error
[*] page fault addr:   	0x0

Processing payload_00005..
ptdump of payload_00005 failed with return code 1. Output:

ERR: 	TNT 13 at position <0xfffff803444d1574,0xfffff8033a02a6b5>
[*] decoder returned: decoder_error
[*] page fault addr:   	0x0

Processing payload_00006..
ptdump of payload_00006 failed with return code 1. Output:

ERR: 	TNT 14 at position <0xfffff803444d1574,0xfffff8033a02a6b5>
[*] decoder returned: decoder_error
[*] page fault addr:   	0x0

```

M3noetius, 2023-09-14:
Of course now I see that when I run `kafl cov` an new instance of qemu is running and the driver is getting loaded on another memory region so thats why i guess the error occurs.

Wenzel, 2023-10-09:
Hi @M3noetius,

yes, when you run `kafl cov`, you should specify the `--resume` paramter, to start from the same kAFL snapshot that was using during the fuzzing campaign.

This isn't well documented at this point, but i'm working on a new Linux kernel tutorial that will better explain these concepts.

Tell me if your issue is solved.