IntelLabs/kAFL #239: Windows WINAPI functions don't seem to work normally in kAFL 0.7.

bjloed, 2023-10-08:
The provided code is an example of creating a log file using WinAPI:
```
void makeLog() {
    HANDLE logHandle = CreateLogFile(L"LOG:C:\\Users\\Public\\Base2",
        GENERIC_WRITE | GENERIC_READ,
        2,
        NULL,
        OPEN_ALWAYS,
        0);

    if (logHandle == INVALID_HANDLE_VALUE) {
        hprintf("CreateLogFile Error\n");
    }
    else {
        hprintf("[makeLog] handle: %x\n", logHandle);
    }
    CloseHandle(logHandle);
```

```
int main(int argc, char** argv)
{
    kAFL_payload* payload_buffer = (kAFL_payload*)VirtualAlloc((kAFL_payload*)0x200000, PAYLOAD_MAX_SIZE, MEM_COMMIT | MEM_RESERVE, PAGE_READWRITE);
    memset(payload_buffer, 0x0, PAYLOAD_MAX_SIZE);
    hprintf("harness payload buffer: %x\n", payload_buffer);

    init_agent_handshake();
    init_panic_handlers();

    /* this hypercall submits the current CR3 value */
    kAFL_hypercall(HYPERCALL_KAFL_SUBMIT_CR3, 0);
    /* submit the guest virtual address of the payload buffer */
    kAFL_hypercall(HYPERCALL_KAFL_GET_PAYLOAD, (UINT64)payload_buffer);

    // Submit PT ranges
    set_ip_range();

    // Snapshot here
    kAFL_hypercall(HYPERCALL_KAFL_NEXT_PAYLOAD, 0);
    /* request new payload (*blocking*) */
    kAFL_hypercall(HYPERCALL_KAFL_ACQUIRE, 0);
     
    // execute Windows Common Log File System WinAPI
    makeLog();
    // and read log data
    // ReadFile()....
	
    /* inform fuzzer about finished fuzzing iteration */
    // Will reset back to start of snapshot here
    kAFL_hypercall(HYPERCALL_KAFL_RELEASE, 0);

    return 0;
}
```
When opening the resulting file, it should contain the default log file data. When you run the makeLog() function directly in the QEMU GUI before running kafl-fuzz, the data is correctly included in the file.
[image]
However, when you execute kafl-fuzz, the file is saved with empty contents when the CreateLogFile() function is called within the makeLog() function. In other words, it behaves abnormally.

Why is this happening? Does file I/O pose any issues during fuzzing? Thanks.

5angjun, 2023-10-08:
i also had stucked in this problem.
i think there is some problem in Qemu PT.

🤯🤯

Wenzel, 2023-10-24:
Hi @bjloed 

> When opening the resulting file, it should contain the default log file data.

I'm not familiar with Windows logging functions. Do you expected a certain header in the log file, after `CreateLogFile()` has been called ?
Otherwise your `makeLog()` simply creates an empty file here.

Assuming there should be a header, have you tried to force flush the disk I/O in the kAFL harness ?
It's possible that the file content remains in a cache buffer, waiting to be flushed at another occasion, and the snapshot is reset too quickly.

bjloed, 2023-10-24:
@Wenzel Hi *^^*
Can I set the time to wait to flush?
In other words, can I change the speed at which I set up the snapshot?
Thank you.

Wenzel, 2023-10-24:
Hi,

Not sure I understand what you need here, but if you are looking for a setting to increase the number of persistent iterations between snapshot restore, you can use `-R` parameter:
https://intellabs.github.io/kAFL/reference/hypercall_api.html#fuzzing-without-snapshot-restore

bjloed, 2023-10-24:
@Wenzel 
Adjusting the number of snapshots resulted in the same results. From what we tested, File I/O and Socket I/O did not seem to work normally when purging. Do you think it's equally a matter of time? How can I solve this problem? Thank you.