IntelLabs/kAFL #241: How can i scale up the Target VM's CPU core?

5angjun, 2023-10-09:
Hello, i'm sangjun.

I really interested in your project.

I want scale up the Target vm's Ram & cpu core.

So i fix some script in kafl/examples/windows_x86_64/Vagrantfile
https://github.com/IntelLabs/kafl.targets/blob/master/windows_x86_64/Vagrantfile

But after vagrant up the vm, the max CPU proccessor limit is 4.

is there Any other way to scale up the number of the VM's CPU core??

However, I can scale up the ram size normally, but I can't scale up the CPU core...

Thanks.

[image]

Wenzel, 2023-10-09:
Hello @5angjun,

> I really interested in your project.

Thank you for your interest and opening issues on the problems that you face, this is a good learning experience for us.

> But after vagrant up the vm, the max CPU proccessor limit is 4.

I tested this myself, first by configuring the `libvirt.cpus` value to `1`, which worked.
And then by testing `5` and `6`.

Looking at the virt-manager/CPU panel, the value is really propogated to libvirt XML domain.

So at this point I really believe this is a limitation of the Windows 10 Entreprise Evaluation image that we are using as a template.

You need to activate it ot unlock all of the features and remove hardware limitations.

Wenzel, 2023-10-09:
Update: I found a solution ! :tada: 

So libvirt has a very dump way of adding more CPUs, by simply adding another `socket` to the hardware.
And Windows 10 might be limited to 4 sockets max.

What you can do is play instead with the CPU topology, specifying how many cores are to be found per socket:
- https://vagrant-libvirt.github.io/vagrant-libvirt/configuration.html#provider-options
- https://libvirt.org/formatdomain.html#cpu-model-and-topology

[image]

Testing with 1 socket and 6 cores:
[image]

[image]

It works !

closed 2023-10-09 by Wenzel