IntelLabs/kAFL #243: Slave has Died issue

5angjun, 2023-10-11:
Hello, I'm sangjun who is very interested in this project.

However, I want to know how to fix some error in Manager & Workers Communication.

I want to make died process to restart new qemu and reconnect fuzzing process when slave has died.

Dying slaves is very critical when i try to fuzzing very long hours ex) over 6hours.

So i think what needs to be impored is to re-engage dead workers in the fuzzing process. 

Is any idea of this??
```
    def wait(self, timeout=None):
        results = []
        r, w, e = select.select(self.clients, (), (), timeout)
        for sock_ready in r:
            if sock_ready == self.listener:
                c = self.listener.accept()
                self.clients.append(c)
                self.clients_seen += 1
            else:
                try:
                    msg = sock_ready.recv_bytes()
                    msg = msgpack.unpackb(msg, strict_map_key=False)
                    results.append((sock_ready, msg))
                except (EOFError, IOError):
                    sock_ready.close()
                    self.clients.remove(sock_ready)
                    self.logger.info("Worker disconnected (remaining %d/%d)." % (len(self.clients)-1, self.clients_seen))
                    if len(self.clients) == 1:
                        raise SystemExit("All Workers exited.")
        return results
     ```

Wenzel, 2023-10-11:
Hi @5angjun,

I think we need to understand why the slaves (or Workers) are dying in the first place ?
That shouldn't happen.

You can get more logging information with [`--log`](https://intellabs.github.io/kAFL/reference/fuzzer_configuration.html#log) and combine it with [`--debug`](https://intellabs.github.io/kAFL/reference/fuzzer_configuration.html#debug) to extract useful debug output.

cc @il-steffen , can we expect dying workers during a fuzzing campaign ? Something I'm missing ?

il-steffen, 2023-10-11:
Worker exit can happen on Qemu segfault or unhandled exception in the worker / mutation logic. The above logic is only to handle the loss of the socket connection, you need to look at why the worker exited..

5angjun, 2023-10-11:
I think the error occured when qemu died.

The last died code is this.
```
    def run_qemu(self):
        self.control.send(b'x')
        self.control.recv(1)
  ```
  
  So i think it  is nice to restart fuzzing campaign when qemu die.

il-steffen, 2023-10-11:
Please have a look why this is happening. In general we want to fix anything that causes workers to die during a fuzzing campaign. 
There are cases where restarting won't help, for instance if the disk is full then Qemu will just exit again on next file/log write.

In some cases there may be Qemu segfault that is not easy to fix, for instance we had bugs related to specific virtio fuzzing harnesses where fixing Qemu did not make much sense. In this case it would make sense to catch + restart the worker. This should be possible from the manager, and then the fuzzing campaign can just continue running.

The manager main loop is here: https://github.com/IntelLabs/kafl.fuzzer/blob/master/kafl_fuzzer/manager/manager.py#L85
We enter this just after launching the workers: https://github.com/IntelLabs/kafl.fuzzer/blob/master/kafl_fuzzer/manager/core.py#L104
The workers are python threads which in turn launch Qemu sub-processes. The threads should abort normally on Qemu communication error or uncatched exceptions, so you should be able to detect and restart the thread with same settings.

With some luck, the socket connection code you referenced above should detect the new worker and the main loop will start dispatching jobs again.

5angjun, 2023-10-11:
This situation appears when allocating a lot of RAM to a vm image and performing parallel fuzzing. In my case, this problem appeared while fuzzing the Windows built-in driver for a long time.

For example, my host computer's RAM size is 80G, but when I allocated 10G of RAM to each vm and fuzzed it with 8 cores, qemu or worker died (there is a high probability that qemu died).

But the manager process is still alive. I am thinking about how to modify the code to revive dead workers in the manager process.

As a person who loves kAFL, I will also think about how to modify kAFL to make it a masterpiece.😀😀😀

Thank

5angjun, 2025-04-20:
I've been reviewing the commit ab1c493, and I noticed that Nyx doesn't seem to account for the possibility of QEMU crashing during execution.

While the current implementation seems to continue fuzzing even after a crash, I believe this approach might silently ignore critical issues

Would it make sense to consider a more robust handling strategy here?

il-steffen, 2025-04-20:
Qemu crashes are usually configuration errors or bugs in Qemu itself. The most common case I've seen is simply running out of disk space due to logging or fuzzer findings filling up the disk. But it can also be rare cases of Qemu itself crashing due to strange VM activity. Imagine a campaign running for days with some 64 guest VMs, we don't really care if one of them dies every other day.

Usually, if you encounter this it makes sense to debug and find out what caused Qemu to crash. But it is difficult to implement a response that satisfies all situations. Should we stop or continue? restart?

Since the situation is relatively rare, the default action to simply let the instance exit seems reasonable to me. The kafl GUI also shows when instances have stopped responding, so I would not say we ignore anything. The user can go and investigate. What more robust error handling do you imagine?

5angjun, 2025-04-21:
Thanks for the detailed explanation.

I totally agree that QEMU crashes are often due to environmental issues like disk space, and they can be relatively rare.
So as you mentioned, trying to recover from them automatically might not always be the best approach and could actually make things more confusing in certain situations.

That said, I noticed that when a worker is marked as "stalled" in the GUI, it's not always clear whether the underlying cause is a QEMU crash, a guest VM hang, the mutation stage taking too long, or a communication issue. Would it be possible to distinguish between these causes more explicitly?

In AFLplusplus, for example, they categorize and handle different error types in their forkserver(I considered AFL's forkserver to be conceptually similar to nyx-qemu in kAFL) logic, as shown here:

https://github.com/AFLplusplus/AFLplusplus/blob/247e8241b4b56520f50846433ac9d10b338e8961/src/afl-forkserver.c#L2025

https://github.com/AFLplusplus/AFLplusplus/blob/247e8241b4b56520f50846433ac9d10b338e8961/src/afl-forkserver.c#L1946

I believe having similar categorization could help with better diagnostics and more effective problem resolution.

il-steffen, 2025-04-21:
If you look at the logic for "stalled", it is simply a check if the status was recently updated. To make it show crashes or longer mutation phase, you would have to modify the kafl-fuzzer worker code to provide additional status updates in those cases. It should be easy to do that for qemu crashes. For the mutation updates the best I can think of is to divide mutations into multiple phases and let them update the status more often. If you look at redqueen and AFL bit manipulations, they already do that a lot.

Simpler fix, just increase your gui timeout for "stalling" screen. Or perhaps make it depend on the execution speed of the instance.