IntelLabs/kAFL #246: Is there a patch or document that can show how to patch kafl agent into the target linux kernel?

liujf628995, 2023-10-20:
And are there explanations about the kafl kernel config parameters?For example:
`
grep KAFL config.vanilla.virtio
CONFIG_TDX_FUZZ_KAFL=y
CONFIG_TDX_FUZZ_KAFL_DETERMINISTIC=y
CONFIG_TDX_FUZZ_KAFL_DEBUGFS=y
# CONFIG_TDX_FUZZ_KAFL_TRACE_LOCATIONS is not set
CONFIG_TDX_FUZZ_KAFL_VIRTIO=y
CONFIG_TDX_FUZZ_KAFL_SKIP_MSR=y
CONFIG_TDX_FUZZ_KAFL_SKIP_CPUID=y
# CONFIG_TDX_FUZZ_KAFL_SKIP_IOAPIC_READS is not set
# CONFIG_TDX_FUZZ_KAFL_SKIP_ACPI_PIO is not set
CONFIG_TDX_FUZZ_KAFL_SKIP_RNG_SEEDING=y
# CONFIG_TDX_FUZZ_KAFL_SKIP_PARAVIRT_REWRITE is not set
CONFIG_TDX_FUZZ_KAFL_VANILLA_PAYLOAD=y
CONFIG_TDX_FUZZ_KAFL_VANILLA_INJECTION_SAMPLE=y
`

Wenzel, 2023-10-20:
Hi @liujf628995,

I totally agree that the currrent Linux Kernel Tutorial lacks documentation on the internals of the kafl-agent, and its configuration options.

In my opinion it doesn't serve it's purpose very well as an introduction to kAFL.
I started an initiative to improve the documentation, but it will take time.

That's why I started to build another Linux based target, more accessible: the DVKM module:
https://github.com/IntelLabs/kafl.targets/pull/31

I'm currently building a detailed tutorial for this, and will be publishing it in a few weeks.

Were you specifically interested by the Linux kernel virtualization interfaces fuzzing (Virtio, MSR, CPUID, IOAPIC, RNG,...), and the new DVKM (Damn Vulnerable Kernel Module) target could suit your needs ?

Wenzel, 2023-10-24:
> Would you want to replace the KAFL Agent with DVKM?Or just add this module to the existing agent?

The exisitng kafl-agent was very specialized to fuzz the linux kernel virtualization communication channels, it's very complicated to understand and maintain, even more for a beginner.

The kafl-agent for DVKM will be minimal, only inserting crash hypercalls at the right locations in the kernel, and the rest will be in userland.

I will keep you up to date on this !

Wenzel, 2023-10-31:
Hi @liujf628995 

The DVKM tutorial has now been merged upstream, and the documentation is online:
https://intellabs.github.io/kAFL/tutorials/linux/dvkm/target.html

I'm eager to receive your feedback on it, what works and what doesn't, what's unclear and what unanswered questions you might have.

liujf628995, 2023-11-03:
Thanks for your detailed tutorial,I'll try it these days and give you feedback later.
> Hi @liujf628995
> 
> The DVKM tutorial has now been merged upstream, and the documentation is online: https://intellabs.github.io/kAFL/tutorials/linux/dvkm/target.html
> 
> I'm eager to receive your feedback on it, what works and what doesn't, what's unclear and what unanswered questions you might have.

Wenzel, 2023-11-03:
Thanks !
Closing this issue for now, feel free to reopen if necessary

closed 2023-11-03 by Wenzel