IntelLabs/kAFL #251: Document Error

5angjun, 2023-10-28:
Hi, i'm sangjun.

I think there is an error in kAFL hypercall API document.
https://intellabs.github.io/kAFL/reference/hypercall_api.html

it should be changed. i guess so.

`kAFL_hypercall(HYPERCALL_KAFL_DUMP_FILE, (uint64_t)&dump_file);`

because QEMU-PT recieve structure format through hypercall args1

```

static void handle_hypercall_kafl_dump_file(struct kvm_run *run,
                                            CPUState       *cpu,
                                            uint64_t        hypercall_arg)
{
    kafl_dump_file_t file_obj;
    char             filename[256] = { 0 };
    char            *host_path     = NULL;
    FILE            *f             = NULL;

    uint64_t vaddr = hypercall_arg;
    memset((void *)&file_obj, 0, sizeof(kafl_dump_file_t));

    if (!read_virtual_memory(vaddr, (uint8_t *)&file_obj, sizeof(kafl_dump_file_t),
                             cpu))
    {
        nyx_error("Failed to read file_obj in %s. Skipping..\n", __func__);
        goto err_out1;
    }

    if (file_obj.file_name_str_ptr != 0) {
        if (!read_virtual_memory(file_obj.file_name_str_ptr, (uint8_t *)filename,
                                 sizeof(filename) - 1, cpu))
        {
            nyx_error("Failed to read file_name_str_ptr in %s. Skipping..\n",
                      __func__);
            goto err_out1;
        }
        filename[sizeof(filename) - 1] = 0;
    }
....
...
...
```

[image]

5angjun, 2023-10-31:
@Wenzel Hi i'm sangjun.

I'm happy to contribute to your project. 

However, it seems that someone other than me was tagged in kafl 0.8 release. The beginning of my name is 5. LOL
🙃🙃

Wenzel, 2023-10-31:
Sorry for the mistake !
It has been fixed just now.

Thanks for contributing to kAFL ! :wave:

5angjun, 2023-10-31:
its okay. Very Thanks!!

PR #254 (2023-10-30): docs: fix hypercall argument for DUMP_FILE
Reported by @sangjun
Fixed #251