IntelLabs/kAFL #252: kAFL 0.7+nyx Sleep() function issue :(

bjloed, 2023-10-28:
Hi, I'm currently using kAFL 0.7
When the Sleep() function includes the harness, the ntoskrnl crashes.
The reason for this is that if I run a harness without Sleep() function, it will work normally, but if I add Sleep() to the code that works normally, it will crash
What do you think the reason is?
Thanks

Wenzel, 2023-10-30:
Hi @bjloed,

thanks for opening an issue.
I have no idea why the `Sleep()` function would cause the Windows kernel to crash inside a harness.
But since you have a crash, you should be able to retrieve the crash dump and perform an initial analysis of the stacktrace.

This should give you a clue what happened.

Wenzel, 2023-10-30:
One more thing:

The Windows driver example locates the `KeBugCheck()` functions and inserts kAFL PANIC events instead.
https://intellabs.github.io/kAFL/tutorials/windows/driver/target.html#panic-handlers

However, Windows will create a crash dump inside the `KeBugCheck()` handler (according to this WRK repo):
https://github.com/mic101/windows/blob/master/WRK-v1.2/base/ntos/ke/bugcheck.c#L1146C15-L1146C15

So when running inside the kAFL harness, if a crash occurs, that `IoWriteCrashDump()` will not be reached.
Keep this in mind while debugging.