IntelLabs/kAFL #259: Qemu USB issue

kbh1860, 2023-11-08:
Hello! I'm interesting to your project.

It can be add USB in the target vm??

I'm add this option in `kafl_fuzzer/common/config/default_settings.yml`

qemu_base: -enable-kvm -machine kAFL64-v1 -cpu kAFL64-Hypervisor-v1,+vmx -no-reboot -display none -net none -device nec-usb-xhci -device usb-host,hostbus=1,hostport=5

like this, but the error has shown "qemu-system-x86_64: -device usb-host,hostbus=1,hostport=5: 'usb-host' is not a valid device model name"

Qemu-nyx doesn't have usb add features?? Or is there another way?

Wenzel, 2023-11-09:
Hi @kbh1860 
thank you for your interest in kAFL !

It seems that `libusb` is disabled by default in QEMU, since our configure line doesn't enable or disable that feature explicitely:
https://github.com/IntelLabs/kafl.qemu/blob/kafl_stable/compile_qemu_nyx.sh#L78

you will need to add `--enable-libusb` to that line, and and instruct kAFL deployment to simply run the [`build`](https://intellabs.github.io/kAFL/reference/deployment.html#makefile-targets) task:

```shell
make deploy -- --tags build
```

Wenzel, 2024-01-26:
Hi @kbh1860 did you solve your issue ?
Can we close it ?