IntelLabs/kAFL #260: Remote virt-viewer and provisioning

amonnymouse, 2023-11-08:
Hi.

I'm now preparing Cloud Fuzzing Settings with kAFL.

But there is some problem like this.
https://github.com/IntelLabs/kafl.targets/blob/075090677a2384dee43d4f5461623f2c6e637e77/windows_x86_64/Makefile#L23

when i provisioning Target Driver, I want to see the remote access when provisioning stage.

i want to see Vm with GUI.

like this. `virt-viewer -a -c qemu:///session vagrant-kafl-windows`
this command can apply just localhost, but remote can't access the vm...

Have any good idea to see the Windows VM with Provisioned Driver in remote access?

liujf628995, 2023-12-21:
The kafl windows use spice for remote handling,I added this param in the Vagrantfile(kAFL/kafl/examples/windows_x86_64/Vagrantfile):

```
libvirt.graphics_port = 5900
```
then you can connect the vm with spice://localhost:5900
The provisioning stage used winrm and I'm not sure it could be viewed in remote window,maybe we can manually operate the command line from this yaml(kAFL/kafl/examples/windows_x86_64/setup_target.yml) in the remote window to debug.

Wenzel, 2024-01-26:
Hi,

> when i provisioning Target Driver, I want to see the Vm with remote access when provisioning stage.

The VM will be managed by libvirt, so you should be able to update the domain definition in vagrant: https://github.com/IntelLabs/kafl.targets/blob/master/windows_x86_64/Vagrantfile#L24

And update the `graphics_ip` to be bound to `0.0.0.0`, and you should be able to connect with the SPICE protocol.
you can also switch to VNC, but it's less reliable (I had erratic mouse mouvments)