IntelLabs/kAFL #261: kAFL cov didn't work??

5angjun, 2023-12-01:
[image]
[image]

I think there is an error in kafl cov functioning.
`kafl cov -w work`

Now, i'm trying to find root cause of this, i think that caused by PT range.

In kafl cov's qemu option is `-device nyx,chardev=nyx_socket,workdir=/home/b/kAFL/kafl/examples/windows_x86_64/work_Process-1,worker_id=1337,bitmap_size=65536,input_buffer_size=131072,dump_pt_trace,ip0_a=0xfffff80025610000,ip0_b=0xfffff8002563f000
`

but the real range is `0xFFFFF8063DDA0000	0xFFFFF8063DDCF000	target.sys
` and libxdc decode failed.
[image]

Wenzel, 2024-01-26:
Hi @5angjun 

If I understand correctly, there is a mismatch between the range used `kafl cov` and the real range of your `target.sys` driver ?

`kafl cov` will start will reloading the Intel PT ranges, if they have been dumped by `kafl fuzz` previously:
https://github.com/IntelLabs/kafl.fuzzer/blob/master/kafl_fuzzer/coverage/__init__.py#L485
dumping config by kafl fuzz:
https://github.com/IntelLabs/kafl.fuzzer/blob/master/kafl_fuzzer/manager/core.py#L129

Do you have a minimal example i can repro and investigate ?

5angjun, 2024-01-27:
yes, there is a mismatch in intelRange.

when booting the qemu with kafl cov,
the target.sys pt range will set below value by dump file.
ip0_a=0xfffff80025610000,ip0_b=0xfffff8002563f000

but after qemu is loaded , the target.sys's mapped different addresss.

5angjun, 2025-04-20:
Since kafl-cov uses the previously stored driver address range, it doesn’t reflect the actual driver base address after the VM reboots. This leads to decoding issues due to mismatched PT ranges.

A patch that dynamically updates the PT range based on the current driver address would likely resolve the issue.

Would it be possible for you to provide a patch that properly handles this issue?

https://github.com/IntelLabs/kAFL/issues/236

5angjun, 2025-04-23:
I successfully recover the edge with removing former ip0 range.

With newly driver address, I can get the block and edge count with 'kafl cov -w work' command.

I will closed this issue

closed 2025-04-23 by 5angjun