IntelLabs/kAFL #264: Example of Fuzzing Windows Usermode Process

hyjun0407, 2024-01-03:
Hi, I was wondering if you have any examples of purging for independent window processes.
Yes) How do I find out if Agent.exe sends a request to test.exe with a specific function and test.exe crashes? I thought of an example of creating VEH through dll injection and sending Hypercall, but it seems too complicated. Is there an example like this?

Wenzel, 2024-01-26:
Hi @hyjun0407 thanks for your interest in kAFL

> Any good example of purging 

purging ?

>  I thought of an example of creating VEH through dll injection and sending panic Hypercall from test.exe's Injected dll, but it seems too complicated.

Could you explain a bit more what you are trying to achieve, and how your target is executing ?

I be happy to integrate a new Windows example to the collection we have.

hyjun0407, 2024-02-03:
I was thinking about how Agent(A) sends a request to another process(B) and PT measures the coverage of the other process(B), which is somewhat possible with the basic example, but what if we send a request from A.exe (Agent in our case) and receive a request from B.exe and mutate based on B's coverage, like in RPC/COM!

closed 2024-02-03 by hyjun0407