IntelLabs/kAFL #265: Fuzzer crashes at "[QEMU-NYX] Booting VM to start fuzzing..."

jjeongxu, 2024-01-04:
Hi,

I really appreciate about the project that you built.
But, when I try to run "kafl fuzz" after setting up the fuzzer and environment(Windows 10 VM, target driver, vuln_test.exe), it crashes after showing up this message
[image]

and the error log varies.
Mostly, it says "assertion failure in 'qemu/nyx/snapshot/block/block_cow.c:450' assert(self->offset_primary < self->cow_primary_size);"
[image]

but sometimes, it says "error no.32 broken pipe"

Command below is the command that I used to run the fuzzer
```
kafl fuzz -w work --seed-dir seed --redqueen -p 1 --t-soft 255 --purge -i result.json --use_call_stack --t-hard 255
```

And the text below is the command that kAFL used to run the VM with QEMU
```
/home/dltkrgksmf/kAFL/kafl/qemu/x86_64-softmmu/qemu-system-x86_64
	-enable-kvm
	-machine kAFL64-v1
	-cpu kAFL64-Hypervisor-v1,+vmx
	-no-reboot
	-net none
	-display none
	-chardev socket,server,id=nyx_socket,path=/home/dltkrgksmf/kAFL/kafl/examples/windows_x86_64/work/interface_0
	-device nyx,chardev=nyx_socket,workdir=/home/dltkrgksmf/kAFL/kafl/examples/windows_x86_64/work,worker_id=0,bitmap_size=65536,input_buffer_size=131072
	-device isa-serial,chardev=kafl_serial
	-chardev file,id=kafl_serial,mux=on,path=/home/dltkrgksmf/kAFL/kafl/examples/windows_x86_64/work/serial_00.log
	-m 8192
	-drive file=/home/dltkrgksmf/.local/share/libvirt/images/windows_x86_64_vagrant-kafl-windows.img
	-monitor unix:/tmp/monitor.sock,server,nowait
	-fast_vm_reload path=/home/dltkrgksmf/kAFL/kafl/examples/windows_x86_64/work/snapshot/,load=off
```

I ensured that drive file in '/home/dltkrgksmf/.local/share/libvirt/images/windows_x86_64_vagrant-kafl-windows.img' is totally flawless. Because it runs fine on Vagrant.
And I also checked that Windows10 VM runs 'test_vuln.exe' when it boots.

[Kernel info]
[image]

[QEMU info]
[image]

Thank you.

Wenzel, 2024-01-05:
Hi @jjeongxu and thanks for the detailed bug report !

I rebuilt the Windows image from scratch and started a new campaign, but couldn't repro the fuzzer or QEMU crashes you experienced.

A couple of things though:

Your `kafl fuzz` command has the following additional parameters
- `-i  results.json`
- `--use_call_stack`

These parameters aren't in the documentation:
https://intellabs.github.io/kAFL/reference/fuzzer_configuration.html

Are you using a kAFL fork by any chance ?

Additionally, regarding the issue of "[QEMU not booting up the VM]", when you boot the VM via kAFL, libvirt isn't involved and can't update its internal state.
if you want to start the VM via libvirt, use virsh API, and tweak the QEMU command line (see [QEMU command line passthrough](https://libvirt.org/kbase/qemu-passthrough-security.html))

jjeongxu, 2024-01-05:
Hi, @Wenzel 
I really appreciate your reply! Thank you for reaching out

I'm using a slightly modified version of kAFL, and the issue was that it couldn't load the target driver into the VM Windows 10's vuln_test process, and there was no single issue with kafl.qemu or kAFL. This resulted in the problem of getting stuck in the infinite loop of vuln_test's main() function( while(1) { CreateFile() } ).

Now, everything is resolved, and I sincerely appreciate your response.

Wishing you a Happy New Year with lots of blessings!

closed 2024-01-05 by jjeongxu

DefenceLabs, 2024-12-05:
Hello, just wanted to know how exactly you fixed this error? Thanks

jjeongxu, 2025-01-11:
> Hello, just wanted to know how exactly you fixed this error? Thanks

Sorry for checking it too late. This was my stupid personal mistake. I couldn't get a HANDLE using CreateFile(), and in my harness, it caused an error because it kept allocating resources in an infinite loop while sending DeviceIoControl() calls. Once I fixed it to properly obtain a HANDLE with CreateFile(), the issue was resolved.