IntelLabs/kAFL #271: Memory Leak(?) problem with usermode harness

hyjun0407, 2024-02-04:
Hi, I looked at the example of the KAFL UserMode harness and used the code below to fuzz Defender. By the way, after some time, QEMU will be shut down due to low memory. (Actual Ram 64gb, Swap file 50gb) This is quite weird. As far as I know, fuzzer is returning qemu to the time it called LOCK Hypercall, and working again, so suppose it's not affected by functions such as memory free or freelibrary.(because it will be restore to when I called Lock Hypercall) And no payload will have more than 500mb of memory usage on the host(on my real compute). Why is this happening? Below is my code.

`#include <stdio.h>
#include <windows.h>
#include "nyx_api.h"
#include "engineboot.h"
#include "openscan.h"
#include "scanreply.h"
#include "streambuffer.h"
#define RSIG_BOOTENGINE                       0x4036
#define RSIG_SCAN_STREAMBUFFER                0x403D
 
 
 
typedef DWORD(*__rsignal)(DWORD Code, PVOID Params, DWORD Size);
 
 
 
#define PAYLOAD_SIZE 128 * 1024
#define PE_CODE_SECTION_NAME ".text"
 
static inline void panic(void){
    kAFL_hypercall(HYPERCALL_KAFL_PANIC, (uintptr_t)0x1);
    while(1){}; /* halt */
}
 
LONG CALLBACK exc_handle(struct _EXCEPTION_POINTERS *ExceptionInfo)
{
    DWORD exception_code = ExceptionInfo->ExceptionRecord->ExceptionCode;
 
    if((exception_code == EXCEPTION_ACCESS_VIOLATION) ||
       (exception_code == EXCEPTION_ILLEGAL_INSTRUCTION) ||
        (exception_code == 0xc0000374) ||
       (exception_code == EXCEPTION_STACK_OVERFLOW) ||
       (exception_code == STATUS_STACK_BUFFER_OVERRUN) ||
       (exception_code == STATUS_FATAL_APP_EXIT))
    {
        panic();
    }
 
    return TRUE;
}
 
 
static DWORD EngineScanCallback(PSCANSTRUCT Scan)
{
    // if (Scan->Flags & SCAN_MEMBERNAME) {
    //     //hprintf("Scanning archive member %s", Scan->VirusName);
    // }
    // if (Scan->Flags & SCAN_FILENAME) {
    //     //hprintf("Scanning@@@@ %s", Scan->FileName);
    // }
    // if (Scan->Flags & SCAN_PACKERSTART) {
    //     //hprintf("Packer %s identified.", Scan->VirusName);
    // }
    // if (Scan->Flags & SCAN_ENCRYPTED) {
    //     //hprintf("File is encrypted.");
    // }
    // if (Scan->Flags & SCAN_CORRUPT) {
    //     //hprintf("File may be corrupt.");
    // }
    // if (Scan->Flags & SCAN_FILETYPE) {
    //     //hprintf("File %s is identified as %s", Scan->FileName, Scan->VirusName);
    // }
    // if (Scan->Flags & 0x08000022) {
    //     //hprintf("Threat %s identified.", Scan->VirusName);
    // }
    // // This may indicate PUA.
    // if ((Scan->Flags & 0x40010000) == 0x40010000) {
    //     //hprintf("Threat %s identified.", Scan->VirusName);
[... 220 lines cut ...]
    ScanParams.ScanReply = &ScanReply;
    ScanReply.EngineScanCallback = EngineScanCallback;
    ScanReply.field_C = 0x7fffffff;
    ScanDescriptor.Read = ReadStream;
    ScanDescriptor.GetSize = GetStreamSize;
    ScanDescriptor.GetName = GetStreamName;
    ScanDescriptor.UserPtr = &ScanDescriptor;
    ScanDescriptor.filename = name;
    ScanDescriptor.fp = fopen(name, "rb");
    ScanDescriptor.filename_w = (wchar_t*)malloc((strlen(name) + 1) * sizeof(wchar_t));
    if (ScanDescriptor.fp == NULL) {
        hprintf("open failed\n");
    }
    fseek(ScanDescriptor.fp, 0, SEEK_END);
    ScanDescriptor.file_size = ftell(ScanDescriptor.fp);
 
    if (prsignal(RSIG_SCAN_STREAMBUFFER, &ScanParams, sizeof(ScanParams)) != 0) {
        hprintf("rsignal failed\n");
    }
    fclose(ScanDescriptor.fp);
    free(ScanDescriptor.filename_w);
 
    kAFL_hypercall(HYPERCALL_KAFL_RELEASE, 0);
    return 0;
}`

Wenzel, 2024-02-05:
Hi @hyjun0407 and thanks for your bug report,

I've never experienced a memory leak before, but i haven't used the harness extensively on Windows as well.

> dunno why but if i paste my code to this issue section directly, then indent make weird. So I paste it to pastebin

Just put your code in a markdown code block, it will keep indention and have syntax highlight:

<details>
  <summary>Your snippet</summary>

```c
#include <stdio.h>
#include <windows.h>
#include "nyx_api.h"
#include "engineboot.h"
#include "openscan.h"
#include "scanreply.h"
#include "streambuffer.h"
#define RSIG_BOOTENGINE                       0x4036
#define RSIG_SCAN_STREAMBUFFER                0x403D
 
 
 
typedef DWORD(*__rsignal)(DWORD Code, PVOID Params, DWORD Size);
 
 
 
#define PAYLOAD_SIZE 128 * 1024
#define PE_CODE_SECTION_NAME ".text"
 
static inline void panic(void){
    kAFL_hypercall(HYPERCALL_KAFL_PANIC, (uintptr_t)0x1);
    while(1){}; /* halt */
}
 
LONG CALLBACK exc_handle(struct _EXCEPTION_POINTERS *ExceptionInfo)
{
    DWORD exception_code = ExceptionInfo->ExceptionRecord->ExceptionCode;
 
    if((exception_code == EXCEPTION_ACCESS_VIOLATION) ||
       (exception_code == EXCEPTION_ILLEGAL_INSTRUCTION) ||
        (exception_code == 0xc0000374) ||
       (exception_code == EXCEPTION_STACK_OVERFLOW) ||
       (exception_code == STATUS_STACK_BUFFER_OVERRUN) ||
       (exception_code == STATUS_FATAL_APP_EXIT))
    {
        panic();
    }
 
    return TRUE;
}
 
 
static DWORD EngineScanCallback(PSCANSTRUCT Scan)
{
    // if (Scan->Flags & SCAN_MEMBERNAME) {
    //     //hprintf("Scanning archive member %s", Scan->VirusName);
    // }
    // if (Scan->Flags & SCAN_FILENAME) {
    //     //hprintf("Scanning@@@@ %s", Scan->FileName);
    // }
    // if (Scan->Flags & SCAN_PACKERSTART) {
    //     //hprintf("Packer %s identified.", Scan->VirusName);
    // }
    // if (Scan->Flags & SCAN_ENCRYPTED) {
    //     //hprintf("File is encrypted.");
    // }
    // if (Scan->Flags & SCAN_CORRUPT) {
    //     //hprintf("File may be corrupt.");
[... 234 lines cut ...]
    ScanDescriptor.GetSize = GetStreamSize;
    ScanDescriptor.GetName = GetStreamName;
    ScanDescriptor.UserPtr = &ScanDescriptor;
    ScanDescriptor.filename = name;
    ScanDescriptor.fp = fopen(name, "rb");
    ScanDescriptor.filename_w = (wchar_t*)malloc((strlen(name) + 1) * sizeof(wchar_t));
    if (ScanDescriptor.fp == NULL) {
        hprintf("open failed\n");
    }
    fseek(ScanDescriptor.fp, 0, SEEK_END);
    ScanDescriptor.file_size = ftell(ScanDescriptor.fp);
 
    if (prsignal(RSIG_SCAN_STREAMBUFFER, &ScanParams, sizeof(ScanParams)) != 0) {
        hprintf("rsignal failed\n");
    }
    fclose(ScanDescriptor.fp);
    free(ScanDescriptor.filename_w);
 
    kAFL_hypercall(HYPERCALL_KAFL_RELEASE, 0);
    return 0;
}
```
</details>

Can you run QEMU monitored by valgrind, and see if you can locate the memory leak in the code ?

hyjun0407, 2024-02-05:
Can U lemme know what is valgrind and how can i find memory leak with that?

hyjun0407, 2024-02-05:
And... there's a lot of memory leaks in the first place, for example:
` kAFL_payload* payload_buffer = (kAFL_payload*)VirtualAlloc(0, host_config.payload_buffer_size, MEM_RESERVE | MEM_COMMIT, PAGE_READWRITE);
`
Payloadbuffer is never freed by harness.
and also for range buffer
`    kAFL_ranges* range_buffer = (kAFL_ranges*)VirtualAlloc(NULL, 0x1000, MEM_COMMIT, PAGE_READWRITE);
    memset(range_buffer, 0xff, 0x1000);`
Other than that, never unlocks after locked tracked section(VirtualLock), etc.. there are a lot of problems with the code. However, kafl returns qemu's status to snapshot when "LOCK" called, so I don't think it'll matter if I don't free or unlock memory. Am I wrong? After release, back to snapshot, memory allocation remains the same?

hyjun0407, 2024-02-07:
> Hi @hyjun0407 and thanks for your bug report,
> 
> I've never experienced a memory leak before, but i haven't used the harness extensively on Windows as well.
> 
> > dunno why but if i paste my code to this issue section directly, then indent make weird. So I paste it to pastebin
> 
> Just put your code in a markdown code block, it will keep indention and have syntax highlight:
> 
> Your snippet
> Can you run QEMU monitored by valgrind, and see if you can locate the memory leak in the code ?

I would appreciate it if you could delete the snippet from this part

Wenzel, 2024-02-07:
@hyjun0407 I've removed your code snippet from my comment, but there is still code present in the multiple edits of your first message.
If you don't want to publish your code, just say it upfront, no need to use a weird indentation excuse.

hyjun0407, 2024-02-07:
> @hyjun0407내 댓글에서 코드 조각을 제거했지만 첫 번째 메시지의 여러 편집 내용에는 여전히 코드가 남아 있습니다. 코드를 게시하고 싶지 않다면 미리 말하세요. 이상한 들여쓰기 변명을 사용할 필요가 없습니다.

Whatever the reason, we apologize for the hassle, and if you look at the first revision, you'll notice that it's not highlighted and is strangely indented. I didn't write the code on my own, I just posted code with my quick thoughts and a coworker asked me to remove it. Whatever the reason, if it came across as rude, I'd like to say it wasn't my intention. I'm so Sorry about it.

hyjun0407, 2024-02-07:
> @hyjun0407 I've removed your code snippet from my comment, but there is still code present in the multiple edits of your first message. If you don't want to publish your code, just say it upfront, no need to use a weird indentation excuse.

Whatever the reason, we apologize for the hassle, and if you look at the first revision, you'll notice that it's not highlighted and is strangely indented. I didn't write the code on my own, I just posted code with my quick thoughts and a coworker asked me to remove it. Whatever the reason, if it came across as rude, I'd like to say it wasn't my intention. I'm so Sorry about it.

p0w1, 2024-04-16:
@hyjun0407 I had a similar problem with the memory consumption of certain VMs. Could you upload the included header files of your code to test it or send it privately if you prefer?