IntelLabs/kAFL #274: AddVectoredExceptionHandler() + selffuzz_test.c(win-user) unexpected behavior

bjloed, 2024-03-11:
Hi. I'm currently testing selffuzz_test.c and the exc_handle function. Currently, panic() has been replaced with *(int *)1 = 0x41; in the if statement of selffuzz_test.c.
```C
static inline void panic(void){
    hprintf("pid: %d\n", GetCurrentProcessId());
    hprintf("PANIC PANIC PANIC PANIC\n");
    kAFL_hypercall(HYPERCALL_KAFL_PANIC, (uintptr_t)0x1);
    while(1){

    }; /* halt */
}
```
And if I add exc_handle, the fuzzing result is as follows:

```bash
pid: 2068
PANIC PANIC PANIC PANIC
[QEMU-NYX] hypercall called!
pid: 2068
PANIC PANIC PANIC PANIC
[QEMU-NYX] hypercall called!
pid: 2068
PANIC PANIC PANIC PANIC
[QEMU-NYX] hypercall called!
pid: 2068
PANIC PANIC PANIC PANIC
[QEMU-NYX] hypercall called!
pid: 2068
PANIC PANIC PANIC PANIC
[QEMU-NYX] hypercall called!
pid: 2068
PANIC PANIC PANIC PANIC
[QEMU-NYX] hypercall called!
pid: 2068
PANIC PANIC PANIC PANIC
[QEMU-NYX] hypercall called!
pid: 2068
PANIC PANIC PANIC PANIC
[QEMU-NYX] hypercall called!
pid: 2068
PANIC PANIC PANIC PANIC
[QEMU-NYX] hypercall called!
pid: 2068
PANIC PANIC PANIC PANIC
[QEMU-NYX] hypercall called!
pid: 2068
PANIC PANIC PANIC PANIC
[QEMU-NYX] hypercall called!
pid: 2068
PANIC PANIC PANIC PANIC
[QEMU-NYX] hypercall called!
pid: 2068
PANIC PANIC PANIC PANIC
[QEMU-NYX] hypercall called!
```
And in kafl gui, only 1 crash appears. Why does this situation repeat itself? Thanks.

closed 2024-03-11 by bjloed

Wenzel, 2024-03-15:
Hi @bjloed 

thanks for opening an issue.
I see you already closed it as completed, did you find a solution or workaround ?
Was it a bug in kAFL that deserves to be fixed ?

Thanks

bjloed, 2024-03-18:
@Wenzel  Ah, I solved it because harness was a 32 bit or 64 bit problem. It's not a kAFL code problem.
Thanks!