IntelLabs/kAFL #275: Can't run kafl fuzz after deployment

GanbaruTobi, 2024-04-02:
No errors while installing.
I tried already the 5... kernel and the 6.0.0 kernel on my own Systems, now I installed an Ubuntu 22.04.4 LTS and ran 

```bash
make deploy
reboot
make deploy (to see if support_test will pass, it does!)
make env
kafl fuzz
```

[image]

To me it seems nyx can't run on the 13th Gen CPU? Maybe there are new mitigations?

Reduced output of cat /proc/cpuinfo
```bash
processor	: 19
vendor_id	: GenuineIntel
cpu family	: 6
model		: 183
model name	: 13th Gen Intel(R) Core(TM) i5-13600K
stepping	: 1
microcode	: 0x11d
cpu MHz		: 1100.005
cache size	: 24576 KB
physical id	: 0
siblings	: 20
core id		: 31
cpu cores	: 14
apicid		: 62
initial apicid	: 62
fpu		: yes
fpu_exception	: yes
cpuid level	: 32
wp		: yes
flags		: fpu vme de pse tsc msr pae mce cx8 apic sep mtrr pge mca cmov pat pse36 clflush dts acpi mmx fxsr sse sse2 ss ht tm pbe syscall nx pdpe1gb rdtscp lm constant_tsc art arch_perfmon pebs bts rep_good nopl xtopology nonstop_tsc cpuid aperfmperf tsc_known_freq pni pclmulqdq dtes64 monitor ds_cpl vmx smx est tm2 ssse3 sdbg fma cx16 xtpr pdcm pcid sse4_1 sse4_2 x2apic movbe popcnt tsc_deadline_timer aes xsave avx f16c rdrand lahf_lm abm 3dnowprefetch cpuid_fault invpcid_single ssbd ibrs ibpb stibp ibrs_enhanced tpr_shadow vnmi flexpriority ept vpid ept_ad fsgsbase tsc_adjust bmi1 avx2 smep bmi2 erms invpcid rdseed adx smap clflushopt clwb intel_pt sha_ni xsaveopt xsavec xgetbv1 xsaves split_lock_detect avx_vnni dtherm ida arat pln pts hwp hwp_notify hwp_act_window hwp_epp hwp_pkg_req hfi umip pku ospke waitpkg gfni vaes vpclmulqdq tme rdpid movdiri movdir64b fsrm md_clear serialize pconfig arch_lbr ibt flush_l1d arch_capabilities
vmx flags	: vnmi preemption_timer posted_intr invvpid ept_x_only ept_ad ept_1gb flexpriority apicv tsc_offset vtpr mtf vapic ept vpid unrestricted_guest vapic_reg vid ple shadow_vmcs ept_mode_based_exec tsc_scaling usr_wait_pause
bugs		: spectre_v1 spectre_v2 spec_store_bypass swapgs eibrs_pbrsb
bogomips	: 6988.80
clflush size	: 64
cache_alignment	: 64
address sizes	: 46 bits physical, 48 bits virtual
power management:
```

GanbaruTobi, 2024-04-02:
[image]

After disabling cpu mitigations via grub (mitigations=off) , it seems to fail at another part:
[image]

GanbaruTobi, 2024-04-02:
Okey, I think it is working now because of mitigations=off
Maybe I will try to disable e-cores if more problems arise (like @schumilo suggested)
[image]

closed 2024-04-02 by GanbaruTobi

Wenzel, 2024-04-04:
Hi @GanbaruTobi,

thank you for this detailed issue, and your investigation regarding `mitigations=off` !
I've opened an issue to track this, and we are working on providing an upgraded kAFL compatible kernel that might solve this problem.

Wenzel, 2024-07-08:
Hi @GanbaruTobi,
I've just released kAFL [v0.9](https://github.com/IntelLabs/kAFL/releases/tag/v0.9), with a Nyx kernel upgraded to 6.8.
I believe this should solve your issue without having to put the `mitigations=off` workaround.

If you could test this kernel and give me a feedback, I would very much appreciate !