IntelLabs/kAFL #281: In the kafl windows target, using the Sleep() function after a snapshot recovery seems to cause the client to stop running

nj00001, 2024-05-09:
Modifying the windows sample program selffuzz_test directly should reproduce the problem

```
    kAFL_hypercall(HYPERCALL_KAFL_NEXT_PAYLOAD, 0);
    kAFL_hypercall(HYPERCALL_KAFL_ACQUIRE, 0);
    
    fuzzme(payload_buffer->data, payload_buffer->size);
    Sleep(1);
    kAFL_hypercall(HYPERCALL_KAFL_RELEASE, 0);
```
[image]

The result is that every fuzz results in a timeout.
And in my tests, many functions with hibernation cause this problem, such as Sleep(), select(), MsgWaitForMultipleObjectsEx(). This is actually quite a serious problem, since many programs use sleep or blocking to synchronize in one way or another.

bjloed, 2024-08-13:
Oh, I met the same bug too. HAHA

closed 2024-08-17 by nj00001

wxzzzzzzz, 2025-12-14:
@nj00001 Have you solved this problem?

wxzzzzzzz, 2025-12-15:
@nj00001 Have you resolved this issue?

nj00001, 2025-12-16:
> [@nj00001](https://github.com/nj00001) Have you resolved this issue?这个问题解决了吗？

When I was organizing old projects, I deleted all commit records. The code in my repository was later re-ported, so some code might be missing. After fixing the issue and restoring the status, I actually fuzzed many UI programs, including the Office suite. Aside from the slower speed, there were no other problems.