IntelLabs/kAFL #288: Issue in running "kafl fuzz".

JoongunPark, 2024-07-25:
Hi, I am trying to follow instruction here (https://intellabs.github.io/kAFL/tutorials/installation.html)

When I run "kafl fuzz" it shows like below.

[image]

So, I ran below commands.

```
touch /dev/shm/kafl_un/aux_buffer_0
truncate -s 4096 /dev/shm/kafl_un/aux_buffer_0
```

Than I got below errors.

```
Warning: Launching without --seed-dir?
No PT trace region defined.
00:00:00:     0 exec/s,    0 edges,  0% favs pending, findings: <0, 0, 0>
Worker-00 Launching virtual machine...
/home/un/kAFL/kafl/qemu/x86_64-softmmu/qemu-system-x86_64
        -enable-kvm
        -machine kAFL64-v1
        -cpu kAFL64-Hypervisor-v1,+vmx
        -no-reboot
        -net none
        -display none
        -chardev socket,server,id=nyx_socket,path=/dev/shm/kafl_un/interface_0
        -device nyx,chardev=nyx_socket,workdir=/dev/shm/kafl_un,worker_id=0,bitmap_size=65536,input_buffer_size=131072
        -device isa-serial,chardev=kafl_serial
        -chardev file,id=kafl_serial,mux=on,path=/dev/shm/kafl_un/serial_00.log
        -m 256
        -append nokaslr oops=panic nopti mitigations=off console=ttyS0
        -fast_vm_reload path=/dev/shm/kafl_un/snapshot/,load=off
[QEMU-NYX] Max Dirty Ring Size -> 1048576 (Entries: 65536)
qemu-system-x86_64: -append only allowed with -kernel option
Worker-00 Shutting down Qemu after 0 execs..
Worker-00 Exit.
Process Worker 0:
Traceback (most recent call last):
  File "/usr/lib/python3.10/multiprocessing/process.py", line 314, in _bootstrap
    self.run()
  File "/usr/lib/python3.10/multiprocessing/process.py", line 108, in run
    self._target(*self._args, **self._kwargs)
  File "/home/un/kAFL/kafl/fuzzer/kafl_fuzzer/worker/worker.py", line 36, in worker_loader
    worker.start()
  File "/home/un/kAFL/kafl/fuzzer/kafl_fuzzer/worker/worker.py", line 130, in start
    if self.q.start():
  File "/home/un/kAFL/kafl/fuzzer/kafl_fuzzer/worker/qemu.py", line 270, in start
    self.__qemu_handshake()
  File "/home/un/kAFL/kafl/fuzzer/kafl_fuzzer/worker/qemu.py", line 301, in __qemu_handshake
    if not self.qemu_aux_buffer.validate_header():
  File "/home/un/kAFL/kafl/fuzzer/kafl_fuzzer/worker/qemu_aux_buffer.py", line 74, in validate_header
    logger.error("Magic mismatch: %x != %x" % (qemu_magic, my_magic))
AttributeError: module 'kafl_fuzzer.common.logger' has no attribute 'error'
Worker disconnected (remaining 0/1).
Manager exit: All Workers exited.
Waiting for Workers to shutdown...

```

Removing all lines with `logger.error` doesn't help. It shows another error like below.
It says "Worker-00 Invalid header in qemu_aux_buffer.py. Abort."

Is there any cleaner way that I can try `kafl fuzz`?
Thanks!

## Evironement
$ uname -a
`Linux un 6.8.0-nyx+ #1 SMP PREEMPT_DYNAMIC Sat Jul  6 13:54:21 UTC 2024 x86_64 x86_64 x86_64 GNU/Linux
`
$ scpu
```
Architecture:            x86_64
  CPU op-mode(s):        32-bit, 64-bit
  Address sizes:         46 bits physical, 57 bits virtual
  Byte Order:            Little Endian
CPU(s):                  144
  On-line CPU(s) list:   0-143
Vendor ID:               GenuineIntel
  Model name:            Genuine Intel(R) CPU $0000%@
    CPU family:          6
    Model:               106
    Thread(s) per core:  2
    Core(s) per socket:  36
    Socket(s):           2
    Stepping:            5
    CPU max MHz:         3600.0000
    CPU min MHz:         800.0000
    BogoMIPS:            4400.00
    Flags:               fpu vme de pse tsc msr pae mce cx8 apic sep mtrr pge mca cmov pat pse36 clflush dts acpi mmx fxsr sse sse2 ss ht tm pbe syscall nx pdpe1gb rdtscp lm constant_tsc art arch_perfmon pebs bts rep_good nopl xtopology nonstop_tsc cpuid aperfmperf pni pc
                         lmulqdq dtes64 monitor ds_cpl vmx smx est tm2 ssse3 sdbg fma cx16 xtpr pdcm pcid dca sse4_1 sse4_2 x2apic movbe popcnt tsc_deadline_timer aes xsave avx f16c rdrand lahf_lm abm 3dnowprefetch cpuid_fault epb cat_l3 intel_ppin ssbd mba ibrs ibpb stib
                         p ibrs_enhanced tpr_shadow flexpriority ept vpid ept_ad fsgsbase tsc_adjust bmi1 avx2 smep bmi2 erms invpcid rtm cqm rdt_a avx512f avx512dq rdseed adx smap avx512ifma clflushopt clwb intel_pt avx512cd sha_ni avx512bw avx512vl xsaveopt xsavec xgetb
                         v1 xsaves cqm_llc cqm_occup_llc cqm_mbm_total cqm_mbm_local split_lock_detect wbnoinvd dtherm ida arat pln pts hfi vnmi avx512vbmi umip pku ospke avx512_vbmi2 gfni vaes vpclmulqdq avx512_vnni avx512_bitalg tme avx512_vpopcntdq la57 rdpid fsrm md_c
                         lear pconfig flush_l1d arch_capabilities
Virtualization features:
  Virtualization:        VT-x
Caches (sum of all):
  L1d:                   3.4 MiB (72 instances)
  L1i:                   2.3 MiB (72 instances)
  L2:                    90 MiB (72 instances)
  L3:                    108 MiB (2 instances)
NUMA:
  NUMA node(s):          2
  NUMA node0 CPU(s):     0-35,72-107
  NUMA node1 CPU(s):     36-71,108-143
Vulnerabilities:
  Gather data sampling:  Vulnerable: No microcode
  Itlb multihit:         Not affected
  L1tf:                  Not affected
  Mds:                   Not affected
  Meltdown:              Not affected
  Mmio stale data:       Vulnerable: Clear CPU buffers attempted, no microcode; SMT vulnerable
  Retbleed:              Not affected
  Spec rstack overflow:  Not affected
  Spec store bypass:     Mitigation; Speculative Store Bypass disabled via prctl
  Spectre v1:            Mitigation; usercopy/swapgs barriers and __user pointer sanitization
  Spectre v2:            Mitigation; Enhanced / Automatic IBRS, IBPB conditional, RSB filling, PBRSB-eIBRS SW sequence
  Srbds:                 Not affected
  Tsx async abort:       Vulnerable: Clear CPU buffers attempted, no microcode; SMT vulnerable

```

5angjun, 2024-07-26:
I think resume options was not implemented after kAFL0.2v.

And `/dev/shm/kafl_un/aux_buffer_0` `/dev/shm/kafl_un/aux_buffer_0` are the file that fuzzer communicate with VM, so you haven't to make that file. I think --resume options make this error.
Can you retry it without resume?

JoongunPark, 2024-07-27:
Thank you for your comment! @5angjun .
Even without the --resume, it shows the same error you can see at the first picture.

JoongunPark, 2024-07-27:
This is result of make deploy

 ```
un  un  ~  kAFL   master  make deploy
make -C deploy deploy --
make[1]: Entering directory '/home/un/kAFL/deploy'
venv/bin/ansible-galaxy install -r requirements.yml --force
Starting galaxy collection install process
Process install dependency map
Starting collection install process
Installing 'intellabs.kafl:1.0.0' to '/home/un/.ansible/collections/ansible_collections/intellabs/kafl'
Created collection for intellabs.kafl:1.0.0 at /home/un/.ansible/collections/ansible_collections/intellabs/kafl
intellabs.kafl:1.0.0 was installed successfully
┏━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━┓
┃            Git repositories will be reset. Uncommitted changes will be lost.          ┃
└━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━┘
┏━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━┓
┃ Ansible BECOME password: if you are using a passwordless SUDO, skip by pressing enter.┃
└━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━┘
venv/bin/ansible-playbook -i /home/un/kAFL/deploy/inventory /home/un/kAFL/deploy/site.yml -K
BECOME password:

PLAY [KAFL deployment] *********************************************************************************************************************************************************************************************************************************************************

TASK [Gathering Facts] *********************************************************************************************************************************************************************************************************************************************************
ok: [localhost]

TASK [Apt update to ensure root access is available (or fail early)] ***********************************************************************************************************************************************************************************************************
changed: [localhost]

TASK [intellabs.kafl.kernel : Checking for intel_pt support in cpuinfo] ********************************************************************************************************************************************************************************************************
ok: [localhost]

TASK [intellabs.kafl.kernel : Checking that pmu_name is not broadwell] *********************************************************************************************************************************************************************************************************
ok: [localhost]

TASK [intellabs.kafl.kernel : Install required dependencies] *******************************************************************************************************************************************************************************************************************
ok: [localhost]

TASK [intellabs.kafl.kernel : Create temporary directory] **********************************************************************************************************************************************************************************************************************
changed: [localhost]

TASK [intellabs.kafl.kernel : Upload support_test.c] ***************************************************************************************************************************************************************************************************************************
changed: [localhost]

TASK [intellabs.kafl.kernel : Compile support_test.c] **************************************************************************************************************************************************************************************************************************
ok: [localhost]

TASK [intellabs.kafl.kernel : Run support_test] ********************************************************************************************************************************************************************************************************************************
ok: [localhost]

TASK [intellabs.kafl.kernel : Remove temp directory] ***************************************************************************************************************************************************************************************************************************
changed: [localhost]

TASK [intellabs.kafl.kernel : Install dependencies] ****************************************************************************************************************************************************************************************************************************
skipping: [localhost]

TASK [intellabs.kafl.kernel : Create temporary directory for downloaded packages] **********************************************************************************************************************************************************************************************
skipping: [localhost]

TASK [intellabs.kafl.kernel : Download deb packages] ***************************************************************************************************************************************************************************************************************************
skipping: [localhost] => (item=https://github.com/IntelLabs/kafl.linux/releases/download/kvm-nyx-6.8/linux-image-6.8.0-nyx+_6.8.0-g0509b850d2bd-1_amd64.deb)
skipping: [localhost]

[... cut ...] **
ok: [localhost]

PLAY RECAP *********************************************************************************************************************************************************************************************************************************************************************
localhost                  : ok=31   changed=6    unreachable=0    failed=0    skipped=27   rescued=0    ignored=0

make[1]: Leaving directory '/home/un/kAFL/deploy'
 un  un  ~  kAFL   master  jobs
```

Wenzel, 2024-07-30:
Hi @JoongunPark 

when you run kafl fuzz right after the installation has completed, the default configuration defines a [`qemu_append`](https://github.com/IntelLabs/kafl.fuzzer/blob/24c8ba1700848e4d08b5285235623f6afaf25796/kafl_fuzzer/common/config/default_settings.yaml#L27) option.

But without specifying any target (kernel, bios, or qcow disk), QEMU will complain with the error message that you saw (append only allowed with -kernel)
So this is expected.
A ticket is opened here in this regard: https://github.com/IntelLabs/kafl.fuzzer/issues/64

Regarding the message `/dev/shm/kafl_un/aux_buffer_0`, since QEMU didn't launch, it couldn't create the aux buffer expected by the kafl fuzzer.
There is no need to create this file.

Although I agree that this output is confusing for the end user.

PR #289 (2024-07-30): tutorial: clarify when kafl fuzz succeeds
Fix #288 

Requires https://github.com/IntelLabs/kafl.fuzzer/releases/tag/v0.6.1
closed 2024-07-30 by Wenzel