IntelLabs/kAFL #290: Why Hyper-V doesn't work in nyx 6.8 kernel?

bjloed, 2024-08-22:
I understand that Hyper-V build flags have been added to the kernel since the kernel came up to 6.8  
However, in the nyx kernel, modified mmu.c and mmu.h, so Hyper-V doesn't seem to work properly. (Can't build)
Is there a reason for the modification? I want to use Hyper-V in nyx 6.8 kernel :(

https://github.com/IntelLabs/kafl.linux/commit/99f6245f99fa3fdd2008624b120aa2d4c7554fcb

Thanks!

Wenzel, 2024-08-23:
Hi @bjloed 

Could you be more specific regarding your build failure with the Nyx kernel sources ?
Which config option did you try to enable ?

Thanks !

bjloed, 2024-08-24:
Hello, @Wenzel  Thanks for your response!

Here's the situation I'm currently facing: First, using the 5.12.x kernel from https://github.com/SB-GC-Labs/hAFL1, I successfully created and booted a Hyper-V VM instance within QEMU without modifying the build options much.

However, with the kAFL 0.7 kernel, which has nyx applied, after applying the same QEMU options and modifying the Vagrantfile, either the host system freezes, or Hyper-V installs within QEMU but the VM instance fails to run, showing a hardware error warning.

In the case of the 6.8 kernel, I get a "cpu->kvm_msr_buf->nmsrs" error when running QEMU, or I encounter an error saying that this kernel does not support Hyper-V.

To resolve this, I tried downloading the 6.8 kernel from kafl.linux and building it with the CONFIG_KVM_HYPERV option enabled, but due to a commit mentioned earlier, there’s a conflict with this build option.

Why is it that only the 5.12.x kernel can boot successfully? The QEMU options I applied are as follows: QEMU: -cpu Skylake-Client,hv_relaxed,hv_spinlocks=0x1fff,hv_vapic,hv_time,+intel-pt,-hypervisor,+vmx -enable-kvm

Thanks!