IntelLabs/kAFL #298: kafl_fuzzer manager issues when running single instance

ducphanduyagentp, 2024-09-24:
I am running the windows userspace example and some strange situation comes up. When I run `kafl fuzz` with multiple instance with `-p`, it works normally. But if I run only one instance, this will happen (it did not when I first run it similarly, and now it happens on another machine of mine on the very first run)

```
(.venv) user@io:~/kAFL/kafl/examples/windows_x86_64$ kafl fuzz --purge 

    __                        __  ___    ________
   / /_____  _________  ___  / / /   |  / ____/ /
  / //_/ _ \/ ___/ __ \/ _ \/ / / /| | / /_  / /
 / ,< /  __/ /  / / / /  __/ / / ___ |/ __/ / /___
/_/|_|\___/_/  /_/ /_/\___/_/ /_/  |_/_/   /_____/
===================================================

<< kAFL Fuzzer >>

Warning: Launching without --seed-dir?
No PT trace region defined.
00:00:00:     0 exec/s,    0 edges,  0% favs pending, findings: <0, 0, 0>
Worker-00 Launching virtual machine...
/home/user/kAFL/kafl/qemu/x86_64-softmmu/qemu-system-x86_64
	-enable-kvm
	-machine kAFL64-v1
	-cpu kAFL64-Hypervisor-v1,+vmx
	-no-reboot
	-net none
	-display none
	-chardev socket,server,id=nyx_socket,path=/dev/shm/kafl_user/interface_0
	-device nyx,chardev=nyx_socket,workdir=/dev/shm/kafl_user,worker_id=0,bitmap_size=65536,input_buffer_size=131072
	-device isa-serial,chardev=kafl_serial
	-chardev file,id=kafl_serial,mux=on,path=/dev/shm/kafl_user/serial_00.log
	-m 4096
	-drive file=/home/user/.local/share/libvirt/images/windows_x86_64_vagrant-kafl-windows.img
	-monitor unix:/tmp/monitor.sock,server,nowait
	-fast_vm_reload path=/dev/shm/kafl_user/snapshot/,load=off
[QEMU-NYX] Max Dirty Ring Size -> 1048576 (Entries: 65536)
qemu-system-x86_64: warning: host doesn't support requested feature: CPUID.07H:EBX.hle [bit 4]
qemu-system-x86_64: warning: host doesn't support requested feature: CPUID.07H:EBX.rtm [bit 11]
[QEMU-NYX] Dirty ring mmap region located at 0x715d10700000
[QEMU-NYX] Warning: Invalid sharedir...
[QEMU-NYX] Booting VM to start fuzzing...
[+] Starting... C:\Users\vagrant\Desktop\bin\userspace\selffuzz_test.exe
[+] Creating snapshot...
[host_config] bitmap sizes = <0x10000,0x1000>
[host_config] payload size = 128KB
[host_config] worker id = 00
[+] Allocating buffer for kAFL_payload struct
[+] Submitting buffer address to hypervisor...
[+] range buffer 85760000...
Worker-00 Entering fuzz loop..
Waiting for Workers to shutdown...
Worker-00 Shutting down Qemu after 0 execs..
qemu-system-x86_64: terminating on signal 15 from pid 5084 (/home/user/kAFL/kafl/.venv/bin/python3)
Still waiting on Worker 0 (pid=5084)..  [hit Ctrl-c to abort..]
Traceback (most recent call last):
  File "/home/user/kAFL/kafl/.venv/bin/kafl", line 8, in <module>
    sys.exit(main())
             ^^^^^^
  File "/home/user/kAFL/kafl/fuzzer/kafl_fuzzer/__main__.py", line 34, in main
    args.func(settings)
  File "/home/user/kAFL/kafl/fuzzer/kafl_fuzzer/manager/core.py", line 108, in start
    manager.loop()
  File "/home/user/kAFL/kafl/fuzzer/kafl_fuzzer/manager/manager.py", line 112, in loop
    self.send_next_task(conn)
  File "/home/user/kAFL/kafl/fuzzer/kafl_fuzzer/manager/manager.py", line 82, in send_next_task
    if mmh3.hash(main_bitmap) == self.empty_hash:
TypeError: argument 1 must be read-only bytes-like object, not 'c_ubyte_Array_65536'

```

I am not familiar with the code base enough or know what information may be helpful to triage this. So please let me know what further information you may need and I will be happy to help out.

Thanks!

rafaelscheel, 2025-02-11:
Thanks! Had that issue when doing the windows example. Your fix solved the problem.

Wenzel, 2025-02-18:
Hi @ducphanduyagentp thanks for the bug report and the PR !

I've investigated this as well, and this is caused by the `mmh3.hash()` whose definition has changed since [`v5.0.0`](https://pypi.org/project/mmh3/5.0.0/) (September 18th 2024)

From
~~~python3
def hash(key: StrHashable, seed: int = 0, signed: bool = True) -> int: ...
~~~
To
~~~python3
def hash(key: Union[bytes, str], seed: int = 0, signed: Any = True) -> int: ...
~~~

The root cause is that kafl.fuzzer's dependencies are not pinned (!)

I'll merge your PR and pin all deps as well.

closed 2025-02-18 by Wenzel

ducphanduyagentp, 2025-02-18:
Nice. Thanks for the acknowledgement!