IntelLabs/kAFL #302: When some workers are STALLED, the value of CurExec is wrong

ycdxsb, 2024-12-27:
There is a problem with kAFL's status update. When a node is displayed as STALLED, CurExec still calculates it. As shown in the figure, the status of worker 1 2 3 is displayed as STALLED. CurExec should be the value of worker 0, not the fixed value of worker 1 2 3 plus the current value of worker 0.
[image]

ycdxsb, 2024-12-27:
At the same time, I also want to understand why the STALLED situation occurs. Does kAFL have some native options, such as timeout settings, to avoid or rescue the worker from the STALLED state?

ycdxsb, 2025-01-09:
It seems that qemu is stuck and not responding, similar to https://github.com/IntelLabs/kAFL/issues/243

Wenzel, 2025-01-10:
Hi @ycdxsb !

Thanks for opening this issue
I can start having a look at it in 2 weeks