IntelLabs/kAFL #305: hi, i am not sure what does host payload size too large means?

uiop7774, 2025-01-30:
so i am currently trying to fuzz nf_table,
i've done the test_script, agent.sh, but when i try to run the kafl, it shows:

[image]

i don't know where i need to start debugging.

Wenzel, 2025-01-30:
Hi @uiop7774 

this is a message that originates from a [`habort`](https://intellabs.github.io/kAFL/reference/hypercall_api.html#habort) call like this:
https://github.com/IntelLabs/kafl.targets/blob/master/linux-user/fs_fuzzer/src/fs_fuzzer.c#L94

You have a mismatch between the payload configured in your harness agent and the size reported by the host fuzzer.
This is a safety check to ensure they match.

The default fuzzer payload size is configured here:
https://github.com/IntelLabs/kafl.fuzzer/blob/master/kafl_fuzzer/common/config/default_settings.yaml#L17

uiop7774, 2025-02-03:
hi @Wenzel  thanks for the reply!!!
i've solved this problem, however i immediately meet another problem.

[image]
in the above snapshot, i've added some debugging output, but the main problem is that the libxdc_init() is not working.
i.e. the "Worker-00 WTF error occurred: Guest ABORT: libxdc_init() has failed ..."

however, i can not find anything that has to do with libxdc_init() in python code, so i figured it's the problem of c code,
anyway then i go check out what does libxdc do, it turns out to be the intel PT technology.

so i begin my debug process:

[image]
i checked my host's /dev/shm/kafl_james/serial_00.log(above image), i can clearly see that my test script is running in the guest, and the agent_init() runs successfully, but the test script crash when it should get payload from the host. 

 then i go check why my DVKM, fs_fuzzer is working correctly but my own test script is failing.
so, after my debugging step, i found that the possible problem is the kernel.

in DVKM, the kernel is 
[image]

in fs_fuzzer, the kernel is
[image]

but for my own test setup, i used my compiled linux kernel(from source)
[image]

i am now wondering whether it is because my kernel is compiled without any libxdc configuration, so that the libxdc_init() is not running, can you confirm my suspicion?

thanks

Wenzel, 2025-02-18:
Hi @uiop7774 
happy to hear that you solved your payload issue ! 🎉 

The [`libxdc_init() has failed`](https://github.com/IntelLabs/kafl.qemu/blob/kafl_stable/nyx/pt.c#L237) is located in QEMU.
Also, the target (kernel image) has nothing to do with libxdc.

I'm not certain as to why libxdc fails, and the library doesn't contain meaningful error messages unfortunately.
Any way i can repro this error reliably ?